bao-project/bao-demos #3: compile error for rpi4

wjhtinger, 2022-03-27:


josecm, 2022-03-27:
Could you please indicate what is the toolchain you are using. Also, did you make any changes to the code or changed the bao-hypervisor's repo tag?

wjhtinger, 2022-03-28:
> Could you please indicate what is the toolchain you are using. Also, did you make any changes to the code or changed the bao-hypervisor's repo tag?

I did not change the code, and below is my toolchain.

./aarch64-none-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./aarch64-none-elf-gcc
COLLECT_LTO_WRAPPER=/home/wangjh/work/gcc-arm-11.2-2022.02-x86_64-aarch64-none-elf/bin/../libexec/gcc/aarch64-none-elf/11.2.1/lto-wrapper
Target: aarch64-none-elf
Configured with: /data/jenkins/workspace/GNU-toolchain/arm-11/src/gcc/configure --target=aarch64-none-elf --prefix=/data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-elf/install// --with-gmp=/data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-elf/host-tools --with-mpfr=/data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-elf/host-tools --with-mpc=/data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-elf/host-tools --with-isl=/data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-elf/host-tools --disable-shared --disable-nls --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-newlib --with-pkgversion='GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.14)' --with-bugurl=https://bugs.linaro.org/
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 11.2.1 20220111 (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.14))

josecm, 2022-03-28:
It seems this was an issue triggered by more recent compiler versions. It is fixed by e221a9c.

closed 2022-09-19 by josecm

wusnga8dn, 2023-04-14:
> PLATFORM=rpi4 DEMO=linux+freertos 使用make to comolie，遇到如下日志报错：
> 
> make[1]: Entering directory '/home/wangjh/work/bao/bao-demos/wrkdir/srcs/bao' 编译源文件src/core/cpu.c 在/home/wangjh/work/bao/bao包含的文件中-demos/wrkdir/srcs/bao/src/core/inc/mem.h:22, 来自/home/wangjh/work/bao/bao-demos/wrkdir/srcs/bao/src/core/inc/cpu .h :24, 来自 /home/wangjh/work/bao/bao-demos/wrkdir/srcs/bao/src/core/cpu.c:17:/home/wangjh/work/bao/bao-demos/wrkdir/ srcs/ bao/src/core/cpu.c: 在函数 'cpu_send_msg': /home/wangjh/work/bao/bao-demos/wrkdir/srcs/bao/src/lib/inc/list.h:52:17 : error : 数组下标 'list_t[0]' 部分位于 'uint8_t[1]' {aka 'unsigned char[1]'} [-Werror=array-bounds] 52 的数组边界之外 if (list- > tail != NULL) *list->tail = node; | ~~~~^~~~~~ 在/home/wangjh/work/bao/bao-demos/wrkdir/srcs/bao/src/core/cpu.c:17 包含的文件中： /home/wangjh/work/bao/bao-demos/wrkdir/srcs / bao/src/core/inc/cpu.h:31:16: 注意：引用 '_cpu_if_base' 31 |外部 uint8_t _cpu_if_base； | ^~~~~~~~~~~~ 在/home/wangjh/work/bao/bao-demos/wrkdir/srcs/bao/src/core/inc/mem.h:22 包含的文件中， 来自/ home /wangjh/work/bao/bao-demos/wrkdir/srcs/bao/src/core/inc/cpu.h:24, 来自/home/wangjh/work/bao/bao-demos/wrkdir/srcs/bao/ src /core/cpu.c:17: /home/wangjh/work/bao/bao-demos/wrkdir/srcs/bao/src/lib/inc/list.h:54:20: 错误：数组下标'list_t [0 ]' 部分位于 'uint8_t[1]' {aka 'unsigned char[1]'} [-Werror=array-bounds] 的数组边界之外
> 
> 我已经将 -Wno-array-bounds 添加到 makefile 以绕过这个问题，但我不确定是否是正确的解决方案。

Hello, my hardware platform is the same as yours, but when I run bao, the information printed on the serial port is as follows. Have you encountered such a problem?
U-Boot 2022.10 (Apr 12 2023 - 01:55:48 -0700)

DRAM: 7.9 GiB

RPI 4 Model B (0xd03114)

Core: 202 devices, 16 uclasses, devicetree: board

MMC: mmcnr@7e300000: 1, emmc2@7e340000: 0

Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1...

In: serial

Out: serial

Err: serial

Net: eth0: ethernet@7d580000

PCIe BRCM: link up, 5.0 Gbps x1 (SSC)

starting usb...

Bus xhci_pci: Register 5000420 NbrPorts 5

Starting the controller

USB XHCI 1.00

scanning bus xhci_pci for devices... 2 USB Device(s) found

        scanning usb for storage devices... 0 Storage Device(s) found

Hit any key to stop autoboot: 0

switch to partitions #0, OK

mmc0 is the current device

Scanning mmc 0:1...

Card did not respond to voltage select! : -110

No EFI system partition

BootOrder not defined

EFI boot manager: Cannot load any image

Card did not respond to voltage select! : -110

MMC Device 2 not found

no mmc device at slot 2

                                                                               

Device 0: unknown device

ethernet@7d580000 Waiting for PHY auto negotiation to complete... TIMEOUT!

bcmgenet: PHY startup failed: -110

missing environment variable: pxeuuid

Retrieving file: pxelinux.cfg/01-e4-5f-01-40-9c-76

ethernet@7d580000 Waiting for PHY auto negotiation to complete... TIMEOUT!

bcmgenet: PHY startup failed: -110

Retrieving file: pxelinux.cfg/00000000

ethernet@7d580000 Waiting for PHY auto negotiation to complete... TIMEOUT!

bcmgenet: PHY startup failed: -110

Retrieving file: pxelinux.cfg/0000000

ethernet@7d580000 Waiting for PHY auto negotiation to complete... TIMEOUT!

bcmgenet: PHY startup failed: -110