bao-project/bao-demos #8: no handler for abort ec = 0x20

mg199, 2022-09-16:
hello,
    After I write a  SPI driver in the code, but while I run fatload, hypervisor report a error.

BAO ERROR: no handler for abort ec = 0x20

Is there anything wrong? or some where need to be configuration?

mg199, 2022-09-19:
it's solved

closed 2022-09-19 by mg199

a-daydream, 2022-10-13:
> 
hello,
        I encounter this problem while running bare-metal  demo. 
        Can you show me your solution to this problem?

mg199, 2022-10-13:
Hello,beacuse I migrate SPI Driver to freertos over baohypervisor, It has some code error. maybe you can share your project details.

a-daydream, 2022-10-14:
> Hello,beacuse I migrate SPI Driver to freertos over baohypervisor, It has some code error. maybe you can share your project details.

I tried to running bare-metal demo [Single-guest Baremetal](https://github.com/bao-project/bao-demos/blob/master/demos/baremetal/README.md) , then encountered that problem.
environment ： PLATFORM is qemu-aarch64-virt, and DEMO= baremetal.

I found it may have some problem in memory mapping.When I  disable mmu in el1, there is a new problem
BAO ERROR: no emulation handler for abort(0x9000030 at 0xd84). 

I want to know where the baremetal.bin is load in uboot and the reason for my probleam. Thanks!

mg199, 2022-10-14:
> I tried to running bare-metal demo [Single-guest Baremetal](https://github.com/bao-project/bao-demos/blob/master/demos/baremetal/README.md) , then encountered that problem. environment ： PLATFORM is [...]

I also encounter this problem before, I config the memory mapping cover the abort address this problem is solved.
my environment is raspberry 4B, I'm not try in qemu.  I think you can open a issue to ask josecm for help