bao-project/bao-demos #10: BAO ERROR: no emulation handler for abort(0x9000030 at 0xd84)

Abbott-Ming, 2022-10-13:
PLATFORM is qemu-aarch64-virt, and DEMO= baremetal, when i make run and input the 'go 50000000' into the u-boot, but output next message:
BAO ERROR: no emulation handler for abort(0x9000030 at 0xd84)

I want to know where the baremetal.bin is load in uboot and the reason for my probleam. Thanks!

xuechou, 2022-11-04:
## question 1
baremetal.bin was staticly linked to bao.bin. And entry of bao.bin is 0x5000 0000 where u-boot jump to.

## question 2
Could you please provide more details about your environment (OS, BAO version, build steps)?

josecm, 2022-12-04:
So sorry for the delay. I was not receiving notifications for this repo's issues.

@Abbott-Ming I think your problem is solved already by b5de4da49aa5e7dce1313c4d96d11d740b34b4a0. We were not tracking a specific tag for the baremetal code. As it evolved the physical addresses it was using diverged from the addresses in the bao-demos configurations.

closed 2022-12-04 by josecm