bao-project/bao-demos #11: TFTP error: 'Access violation'

fangkaix, 2022-11-04:
I tried this Demo under WSL2(ubuntu-16.04) ,linux may not booted by u-boot?

here is my env setting:
`export PLATFORM=qemu-aarch64-virt`    
`export DEMO=linux+freertos`

here is booting log:
`char device redirected to /dev/pts/4 (label serial3)
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.4(release):7d31f57
NOTICE:  BL1: Built : 13:15:28, Nov  1 2022
NOTICE:  BL1: Booting BL2
NOTICE:  BL2: v2.4(release):7d31f57
NOTICE:  BL2: Built : 13:15:29, Nov  1 2022
NOTICE:  BL1: Booting BL31
NOTICE:  BL31: v2.4(release):7d31f57
NOTICE:  BL31: Built : 13:15:31, Nov  1 2022

U-Boot 2021.01 (Nov 04 2022 - 15:47:58 +0800)

DRAM:  4 GiB
Flash: 64 MiB
*** Warning - bad CRC, using default environment

In:    pl011@9000000
Out:   pl011@9000000
Err:   pl011@9000000
Net:   eth0: virtio-net#31
Hit any key to stop autoboot:  0 
starting USB...
No working controllers found
USB is stopped. Please issue 'usb start' first.
scanning bus for devices...

Device 0: unknown device

Device 0: unknown device
starting USB...
No working controllers found
BOOTP broadcast 1
DHCP client bound to address 172.22.245.15 (2 ms)
Using virtio-net#31 device
TFTP from server 172.22.245.2; our IP address is 172.22.245.15
Filename 'boot.scr.uimg'.
Load address: 0x40200000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
BOOTP broadcast 1
DHCP client bound to address 172.22.245.15 (1 ms)
Using virtio-net#31 device
TFTP from server 172.22.245.2; our IP address is 172.22.245.15
Filename 'boot.scr.uimg'.
Load address: 0x40400000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...`

fangkaix, 2022-11-07:
qemu current command below:    
run: qemu platform
	$(call print-instructions, $(instuctions), 1, false)
	@$(qemu_cmd) -nographic\
		-M virt,secure=on,virtualization=on,gic-version=3 \
		-cpu cortex-a53 -smp 4 -m 4G\
		-bios $(atf-fip)\
		-device loader,file="$(bao_image)",addr=0x50000000,force-raw=on\
		-device virtio-net-device,netdev=net0\
		-netdev user,id=net0,net=172.22.246.0/24,hostfwd=tcp:127.0.0.1:5555-:22\
		-device virtio-serial-device -chardev pty,id=serial3 -device virtconsole,chardev=serial3

fangkaix, 2022-11-07:
my host ip address below  :     
eth0      Link encap:Ethernet  HWaddr 00:15:5d:02:10:68  
          inet addr:172.22.246.45  Bcast:172.22.255.255  Mask:255.255.240.0
          inet6 addr: fe80::215:5dff:fe02:1068/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2721 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1145 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1192154 (1.1 MB)  TX bytes:235765 (235.7 KB)

josecm, 2022-12-04:
@fangkaix Sorry for the delay. I was not receiving notifications for this repo.

Could you clarify what you mean by:

> u-boot log show tftp access violation

Could you show us the log itself?

> is there any way to boot directly or locally?

If you mean booting directly from TF-A to bao, bypassing u-boot, in this platform (i.e. qemu) I would say yes. We can always load all needed images using the loader device. Then we just need to jump directly to Bao from TF-A However, this might involve some hacky modifications to TF-A, or at least using a small blob in TF-A's fip in place of U-boot. Let me know if you are interested in following this route. I'd be happy to give you all the details.

In real platforms some platforms it might difficult. It all depends on the size of your guest binaries. Although Bao is small enough, large guest's might not fit in a platform's roms, and there will always be the need to load it from some secondary storage using a bootloader before Bao running.

fangkaix, 2022-12-21:
> @fangkaix Sorry for the delay. I was not receiving notifications for this repo.
>
> Could you clarify what you mean by:
>
>
> Could you show us the log itself?
>
>
> If you mean booting directly from TF-A to [...]

I see, thanks for your reply.

I was so interested in your project that I cloned it into my "wsl2" and tried to run the demo, all following the README. But I couldn't get the kernel started.

Thank you so much for your time.

I will show you the log below:
`root@fangkaixin:/usr/local/wsl_workspace/hypervisor_space/bao-demos# make PLATFORM=qemu-aarch64-virt DEMO=linux+freertos run
--------------------------------------------------------------------------------

5) Setup connections and jump to Bao

Qemu will let you know in which pseudo-terminals it placed the virtio serial.
For example:

    char device redirected to /dev/pts/4 (label serial3)

If you are running more than one guest, open a new terminal and connect to it.
For example:

    screen /dev/pts/4

If you are running a Linux guest you can also connect via ssh to it by opening
a new terminal and running:

    ssh root@localhost -p 5555

Finally, make u-boot jump to where the bao image was loaded:

    go 0x50000000

When you want to leave QEMU press Ctrl-a then x.

char device redirected to /dev/pts/7 (label serial3)
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.4(release):7d31f57
NOTICE:  BL1: Built : 13:15:28, Nov  1 2022
NOTICE:  BL1: Booting BL2
NOTICE:  BL2: v2.4(release):7d31f57
NOTICE:  BL2: Built : 13:15:29, Nov  1 2022
NOTICE:  BL1: Booting BL31
NOTICE:  BL31: v2.4(release):7d31f57
NOTICE:  BL31: Built : 13:15:31, Nov  1 2022

U-Boot 2021.01 (Nov 04 2022 - 15:47:58 +0800)

DRAM:  4 GiB
Flash: 64 MiB
*** Warning - bad CRC, using default environment

In:    pl011@9000000
Out:   pl011@9000000
Err:   pl011@9000000
Net:   eth0: virtio-net#31
Hit any key to stop autoboot:  0 
starting USB...
No working controllers found
USB is stopped. Please issue 'usb start' first.
scanning bus for devices...

Device 0: unknown device

Device 0: unknown device
starting USB...
No working controllers found
BOOTP broadcast 1
DHCP client bound to address 172.22.246.15 (2 ms)
Using virtio-net#31 device
TFTP from server 172.22.246.2; our IP address is 172.22.246.15
Filename 'boot.scr.uimg'.
Load address: 0x40200000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
BOOTP broadcast 1
DHCP client bound to address 172.22.246.15 (1 ms)
Using virtio-net#31 device
TFTP from server 172.22.246.2; our IP address is 172.22.246.15
Filename 'boot.scr.uimg'.
Load address: 0x40400000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
=> boot
starting USB...
No working controllers found
USB is stopped. Please issue 'usb start' first.

Device 0: unknown device

Device 0: unknown device
starting USB...
No working controllers found
BOOTP broadcast 1
DHCP client bound to address 172.22.246.15 (0 ms)
Using virtio-net#31 device
TFTP from server 172.22.246.2; our IP address is 172.22.246.15
Filename 'boot.scr.uimg'.
Load address: 0x40200000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
BOOTP broadcast 1
DHCP client bound to address 172.22.246.15 (0 ms)
Using virtio-net#31 device
TFTP from server 172.22.246.2; our IP address is 172.22.246.15
Filename 'boot.scr.uimg'.
Load address: 0x40400000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
=> `

fangkaix, 2022-12-21:
I guess this is caused by the inability to find the kernel image through tftp, but I don't know how to fix it

josecm, 2023-01-05:
@fangkaix when reaching u-boot's `=>` prompt you only need to run the `go 0x50000000` command (not `boot` or anything else). All other images were previously loaded by qemu's loader device (i.e. `-device loader,file="$(bao_image)",addr=0x50000000,force-raw=on`)

josecm, 2023-04-20:
@fangkaix was this issue solved for you?

fangkaix, 2023-05-06:
Oh, yes, thank you. I guess I got it wrong

closed 2023-05-06 by fangkaix