bao-project/bao-demos #12: Can the bao project should run at Raspberry Pie 3b with arm A53

jq12, 2022-12-01:
as the title

josecm, 2022-12-04:
@jq12 the short answer is yes. Since the RPi 3b features a quad-core Cortex-A53 Bao will run on it. 

The main question here is how does the memory configuration and boot processor differ from the RPi 4. If only the memory differs, only a small modification is needed to the [platform descrption file](https://github.com/bao-project/bao-hypervisor/blob/latest/src/platform/rpi4/rpi4_desc.c). In the case the boot processor differs, the firmware and it's configuration used in this demo might need to be adapted.

Unfortunately, our team does not have a RPi 3b available. But in case you want to try this for yourself, I'll keep this issue open to assist you through the process.

josecm, 2023-04-20:
@jq12 were you able to achieve the port to RPI3b?

danielRep, 2023-05-11:
Closed since the issue is staled for quite some time

closed 2023-05-11 by danielRep