bao-project/bao-demos #17: SSH and SCP not working on linux+freertos demo

simonevallifuoco, 2023-02-12:
Hi, I am running Bao with dual-guest linux+freertos configuration on Qemu; I am trying to use SCP to copy a file to the linux guest, but I am getting this output: 
[image]

When trying to connect via SSH this is what i get:
[image]

I am making some mistake? What should I do?

josecm, 2023-02-13:
@simonevallifuoco In my current setup, I am not able to replicate the issue. I'll look into it and get back to you.

simonevallifuoco, 2023-02-13:
Thank you @josecm.
This is what happens when I try to ping the linux guest
[image]

Instead, this is what happens when I try to ping the host from the linux guest:
[image]

simonevallifuoco, 2023-02-13:
Ok, looks like it's working now. In /etc/network/interfaces file eth0 was missing. I added it and run ifup eth0 command

josecm, 2023-02-13:
> Thank you @josecm. This is what happens when I try to ping the linux guest [image]
> 
> Instead, this is what happens when I try to ping the host from the linux guest: [image]

I'm not sure if you can route ICMP through [QEMU's user networking](https://wiki.qemu.org/Documentation/Networking#User_Networking_.28SLIRP.29).

> Ok, looks like it's working now. In /etc/network/interfaces eth0 was missing. I added it and run ifup eth0 command

I'm not able to replicate this. But natively on my machine and on a fresh Ubuntu VM eth0 always comes up during boot. Please update this thread if you identify the reason why it is failing on your setup. I'll do the same if I'm able to replicate the issue (and hopefully solve it).

closed 2023-02-13 by josecm