bao-project/bao-demos #25: Jetson tx2, unable to load bao normally

wusnga8dn, 2023-05-04:
I'm trying to deploy bao to jetson tx2 recently. After I completed the operation in bao-demo, I found that when I restarted tx2 to load bao, I found that tx2 would directly enter the system that comes with tx2, and could not load bao normally. No information was printed on the screen, and I could not I get the u-boot prompt, what is the problem?

danielRep, 2023-05-04:
How are you connecting to the TX2? Don't forget you need to connect through a USB-to-TTL adapter as mentioned [here](https://github.com/bao-project/bao-demos/tree/master/platforms/tx2#5-setup-board). The serial output is through the J21 header, on pins 6(GND), 8(TXD), 10(RXD).

wusnga8dn, 2023-05-05:
> 

> How are you connecting to the TX2? Don't forget you need to connect through a USB-to-TTL adapter as mentioned [here](https://github.com/bao-project/bao-demos/tree/master/platforms/tx2#5-setup-board). The serial output is through the J21 header, on pins 6(GND), 8(TXD), 10(RXD).

Thank you very much for your reply, now I have successfully connected tx2 with the computer. This is the prompt information printed by the serial port:
Tegra186 (P2771-0000-500) # fatload mmc 1 0xa0000000 bao.bin; go 0xa0000000
50012160 bytes read in 5406 ms (8.8 MiB/s)
## Starting application at 0xA0000000  …
Bao Hypervisor
BAO WARNING: smmuv2 does not support coherent page table walks
Bao FreeRTOS guest
Task1: 0
Task2: 0
Task1: 1
Task2: 1
Is this correct please?
I have a problem connecting linux guest with ssh: I use ethernet cable to connect tx2 and computer (ubuntu virtual machine) and set static ip192.168.42.2 for wired connection. But it keeps failing to show on my Ubuntu virtual machine Wired Connected, when I press the rst key to restart the tx2 built-in system, the wired connection can be displayed normally, what is the problem?

This is the system that runs tx2
[image]

This is running bao
[image]

wusnga8dn, 2023-05-06:
> How are you connecting to the TX2? Don't forget you need to connect through a USB-to-TTL adapter as mentioned [here](https://github.com/bao-project/bao-demos/tree/master/platforms/tx2#5-setup-board). The serial output is through the J21 header, on pins 6(GND), 8(TXD), 10(RXD), and NOT the J28 micro-USB port

How can I use ssh to connect to linux guest on tx2?

danielRep, 2023-05-11:
> How can I use ssh to connect to linux guest on tx2?

Can you share the config file that you are using on bao and describe the overall steps on how you are building ubuntu.

wusnga8dn, 2023-05-11:
> Can you share the config file that you are using on bao and describe the overall steps on how you are building ubuntu?

I'm very sorry I don't quite understand what you mean, I didn't build ubuntu with modifying configuration files. I deployed Dual-guest Linux+FreeRTOS on bao, and the running platform is jetson tx2. When I deploy successfully, I plan to use the Ethernet cable to connect the built linux, but when I use the Ethernet cable to connect the tx2 and the Ubuntu virtual machine (my physical host), I find that the ubuntu virtual machine cannot detect the Ethernet cable, so it cannot use the Ethernet The network cable is connected to the linux guest, how to solve this problem?

wusnga8dn, 2023-05-11:
> > > > > How are you connecting to the TX2? Don't forget you need to connect through a USB-to-TTL adapter as mentioned [here](https://github.com/bao-project/bao-demos/tree/master/platforms/tx2#5-setup-board). The serial output is through the J21 header, on pins 6(GND), 8(TXD), 10(RXD), and NOT the J28 micro-USB port
> > > > 
> > > > 
> > > > How can I use ssh to connect to linux guest on tx2?
> > > 
> > > 
> > > Can you share the config file that you are using on bao and describe the overall steps on how you are building ubuntu?
> > 
> > 
> > I'm very sorry I don't quite understand what you mean, I didn't build ubuntu with modifying configuration files. I deployed Dual-guest Linux+FreeRTOS on bao, and the running platform is jetson tx2. When I deploy successfully, I plan to use the Ethernet cable to connect the built linux, but when I use the Ethernet cable to connect the tx2 and the Ubuntu virtual machine (my physical host), I find that the ubuntu virtual machine cannot detect the Ethernet cable, so it cannot use the Ethernet The network cable is connected to the linux guest, how to solve this problem?
> 
> So you're connecting the TX2 to your computer via the ethernet cable right? After, did you change your wired connection on you host ubuntu to this configuration? Address: 192.168.42.2 Netmask: 255.255.255.0
Yes I have set the ethernet line URL as: 192.168.42.2.
Because I used the Raspberry Pi to deploy bao and successfully connected to Linux with an Ethernet cable, I saved the configuration of the Ethernet cable. But when I use tx2 to connect, I find that my Ubuntu host cannot detect the Ethernet cable

danielRep, 2023-05-11:
That's really strange. I've tried again, and its working on my side. 
Try to install the JetPack from NVIDIA and see if you can connect by shh to the linux for tegra.

josecm, 2023-06-28:
@wusnga8dn were you able to run the tx2 demo?

sandro2pinto, 2023-07-13:
No feedback since 2 weeks.

Closing the issue.

closed 2023-07-13 by sandro2pinto