bao-project/bao-demos #30: Linux+freeRTOS for rpi4 - Error: junk at end of line, first unrecognized character is `1'

landgraf, 2023-05-26:
Tried to build using arm-gnu-toolchain-11.3.rel1-x86_64-aarch64-none-linux-gnu and arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-linux-gnu. For whatever reason "linux" expands as "1" and fails to build

```
>>>   Executing post-image script board/qemu/post-image.sh
make[1]: Leaving directory '/mnt/builds/bao/sources/bao-demos/wrkdir/srcs/buildroot-aarch64-v6.1'
mv /mnt/builds/bao/sources/bao-demos/wrkdir/srcs/buildroot-aarch64-v6.1/output/images/*Image /mnt/builds/bao/sources/bao-demos/wrkdir/srcs/buildroot-aarch64-v6.1/output/images/Image-rpi4
dtc /mnt/builds/bao/sources/bao-demos/demos/linux+freertos/devicetrees/rpi4/linux.dts > /mnt/builds/bao/sources/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux.dtb
/mnt/builds/bao/sources/bao-demos/demos/linux+freertos/devicetrees/rpi4/linux.dts:48.3-10: Warning (ranges_format): /reserved-memory:ranges: empty "ranges" property but its #size-cells (1) differs from / (2)
make -C /mnt/builds/bao/sources/bao-demos/guests/linux/lloader ARCH=aarch64 IMAGE=/mnt/builds/bao/sources/bao-demos/wrkdir/srcs/buildroot-aarch64-v6.1/output/images/Image-rpi4 DTB=/mnt/builds/bao/sources/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux.dtb TARGET=/mnt/builds/bao/sources/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux
make[1]: Entering directory '/mnt/builds/bao/sources/bao-demos/guests/linux/lloader'
/opt/arm_cross/arm-gnu-toolchain-11.3.rel1-x86_64-aarch64-none-linux-gnu/bin/aarch64-none-linux-gnu-gcc -Wl,-build-id=none -nostdlib -T loader_aarch64.ld\
	-o /mnt/builds/bao/sources/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux.elf -mcmodel=large  aarch64.S -I. -D IMAGE=/mnt/builds/bao/sources/bao-demos/wrkdir/srcs/buildroot-aarch64-v6.1/output/images/Image-rpi4 -D DTB=/mnt/builds/bao/sources/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux.dtb
aarch64.S: Assembler messages:
aarch64.S:32: Error: junk at end of line, first unrecognized character is `1'
aarch64.S:36: Error: file not found: /mnt/builds/bao/sources/bao-demos/wrkdir/imgs/rpi4/1+freertos/1.dtb
make[1]: *** [Makefile:34: /mnt/builds/bao/sources/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux.elf] Error 1

```

josecm, 2023-06-28:
@landgraf, sorry for the delay in the response, and thanks for raising this issue. 

I believe the issue here might be due to the fact you are using the "GNU/Linux target (aarch64-none-linux-gnu)" toolchain instead of the "bare-metal target (aarch64-none-elf)" one.  Could you check if switching the toolchain solves your issue? Or if you have come up with some solution, describe it here?

sandro2pinto, 2023-07-13:
@landgraf any feedback?

landgraf, 2023-07-14:
> @landgraf, sorry for the delay in the response, and thanks for raising this issue.
>
> I believe the issue here might be due to the fact you are using the "GNU/Linux target (aarch64-none-linux-gnu)" [...]

gcc binary is not part of the aarch64-none-elf but make expected to see one:
```
$ make -j1
make -C /mnt/builds/bao/bao-demos/guests/linux/lloader ARCH=aarch64 IMAGE=/mnt/builds/bao/bao-demos/wrkdir/srcs/buildroot-aarch64-v6.1/output/images/Image-rpi4 DTB=/mnt/builds/bao/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux.dtb TARGET=/mnt/builds/bao/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux
make[1]: Entering directory '/mnt/builds/bao/bao-demos/guests/linux/lloader'
/mnt/builds/bao/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/aarch64-none-elf/bin/gcc -Wl,-build-id=none -nostdlib -T loader_aarch64.ld\
	-o /mnt/builds/bao/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux.elf -mcmodel=large  aarch64.S -I. -D IMAGE=/mnt/builds/bao/bao-demos/wrkdir/srcs/buildroot-aarch64-v6.1/output/images/Image-rpi4 -D DTB=/mnt/builds/bao/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux.dtb
make[1]: /mnt/builds/bao/arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/aarch64-none-elf/bin/gcc: No such file or directory
make[1]: *** [Makefile:34: /mnt/builds/bao/bao-demos/wrkdir/imgs/rpi4/linux+freertos/linux.elf] Error 127
```

josecm, 2023-10-21:
@landgraf Once again, sorry for the (huge) delay. I can't tell what is happening, as I'm sure that makefile uses the full triplet binary name (i.e., aarch64-none-elf-gcc) as long as the environment as the correctly defined `CROSS_COMPILE` either with the */path/to/toolchain/bin/aarch64-none-elf* or just *aarch64-none-elf* if the toolchains bin directory is in your `PATH` environment variable. Actually, [the makefile just assumes the latter](https://github.com/bao-project/bao-demos/blob/d0d6b67a7b9942260a19282857f8d565ea887b35/guests/linux/lloader/Makefile#L11) if `CROSS_COMPILED` is not defined in the environment or passed as a make argument.

bohuazou, 2024-11-25:
I used an older version of Arm GNU toolchain ([gcc-arm-11.2-2022.02-x86_64-aarch64-none-elf](https://developer.arm.com/-/media/Files/downloads/gnu/11.2-2022.02/binrel/gcc-arm-11.2-2022.02-x86_64-aarch64-none-elf.tar.xz?rev=981d8f7e91864070a466d852589598e2&hash=9FCB95F9CCCBE04D2077DC1D47113061C03176F7)), and it worked.