bao-project/bao-demos #34: Linux+Zephyr for rpi4

clachan, 2023-07-11:
Hi,

I wonder what is the main issue of Linux+Zephyr for rpi4?

Thanks,
Clay

sandro2pinto, 2023-07-11:
Hi,

There is no big issue (apart from the fact Zephyr do not officially suport RPI4 - but this is not a big deal).

It is more a question of allocating HR to write a device tree and a bit of integration (and prepare de demo).

Thanks
Sandro

clachan, 2023-07-12:
Thanks, @sandro2pinto. We plan to investigate and see what we can do about this.

Thanks,
Clay

sandro2pinto, 2023-07-12:
Very nice @clachan! Please keep us posted and if you find any roadblocker let us know.

Thanks
Sandro

danielRep, 2023-09-13:
I'll reopen the issue if anything comes up @clachan .

josecm, 2023-10-21:
This was the correct PR to solve @yanjiew1! Thanks again.

closed 2023-10-21 by josecm