bao-project/bao-demos #35: Out-of-bounds Error

e-nieto, 2023-07-12:
While trying to compile the Linux+FreeRTOS demo for RPi4, throws the following error:

```make -C ~/Workspace/Bao-hypervisor/bao-demos/wrkdir/srcs/arm-trusted-firmware-aarch64 bl31 PLAT=rpi4 
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '~/Workspace/Bao-hypervisor/bao-demos/wrkdir/srcs/arm-trusted-firmware-aarch64'
  CC      plat/rpi/common/rpi3_pm.c
In file included from plat/rpi/common/rpi3_pm.c:14:
In function 'mmio_write_64',
    inlined from 'rpi3_pwr_down_wfi' at plat/rpi/common/rpi3_pm.c:182:3:
include/lib/mmio.h:51:35: error: array subscript 0 is outside array bounds of 'volatile uint64_t[0]' {aka 'volatile long unsigned int[]'} [-Werror=array-bounds]
   51 |         *(volatile uint64_t*)addr = value;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
plat/rpi/common/rpi3_pm.c: In function 'plat_setup_psci_ops':
plat/rpi/common/rpi3_pm.c:270:9: error: array subscript 0 is outside array bounds of 'uintptr_t[0]' {aka 'long unsigned int[]'} [-Werror=array-bounds]
  270 |         *entrypoint = sec_entrypoint;
      |         ^~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:1341: ~/Workspace/Bao-hypervisor/bao-demos/wrkdir/srcs/arm-trusted-firmware-aarch64/build/rpi4/release/bl31/rpi3_pm.o] Error 1
make[1]: Leaving directory '~/Workspace/Bao-hypervisor/bao-demos/wrkdir/srcs/arm-trusted-firmware-aarch64'
make: *** [~/Workspace/Bao-hypervisor/bao-demos/platforms/rpi4/make.mk:18: ~/Workspace/Bao-hypervisor/bao-demos/wrkdir/imgs/rpi4/bl31.bin] Error 2
```

I followed the compiling guide on README.md on bao-demos, commit e6f1f6f (at the time of this PR writing, origin/master), up to the point of  `make -j$(nproc)`. 

I am using the following versions of Appendix III programs:
| Program                       | Version    |
|:--------------------------------|:---------------|
| aarch64-none-elf-gcc | 12.2.Rel1 |
| make                             | 4.3            |
| dtc  | 1.6.1 |
| gcc  | 12.3.0-5 |
| mkimage | 2023.01 |
| cmake | 3.26.4 |
| ninja | 1.11.1 |

I am compiling on a Debian system following Testing, x86_64 architecture.

e-nieto, 2023-07-12:
Update: It does also not work when trying to compile for `DEMO=baremetal`, so seems to be a problem of the RPi4 files.

ESCristiano, 2023-07-12:
Hi @e-nieto 

I was able to reproduce your build error, however, the versions of the tools you are using are not compliant/compatible with the tools supported by bao-demos, [Appendix III](https://github.com/bao-project/bao-demos#appendix-iii).

To resolve this issue, please consider downgrading your toolchain `aarch64-none-elf-gcc` to the version specified in [Appendix III](https://github.com/bao-project/bao-demos#appendix-iii), which is 11.3, and everything should compile as expected.

josecm, 2023-07-12:
I believe this was an issue related to arm-trusted-firmware. I've updated the bao arm-trusted -firmware repo to the latest 2.9 version, which already supports GCC 12.1. 

Please delete arm-trusted-firmware from wrkdir/srcs and retry the build. It should work.

e-nieto, 2023-07-12:
Uh I got something interesting...

1. I have deleted everything with `make distclean`, changed CROSS_COMPILE to 11.3, compiled. No problems.
2. I read @josecm comment and though "well let's give it a try". Executed `make distclean`. All tidy.
3. I change CROSS_COMPILE to 12.2, execute `make -j12`, _It works without problem_.

I might have done something wrong, I'll try to reproduce tomorrow.

Thanks for the quick responses though!

e-nieto, 2023-07-12:
Looks like it does. I'll try again tomorrow. Cheers!

e-nieto, 2023-07-13:
So I tried a clean make with both 12.2 and 11.3, and it seems like it works without the need of doing anything. I do not know if it was @josecm that pushed something, but it's fixed now :)

Thank you team!

closed 2023-07-13 by sandro2pinto