bao-project/bao-demos #37: No Output for riscv

GooTal, 2023-07-20:
Hi, i tried to run demo on riscv. Here\`s my commands, following the guide.
```
export CROSS_COMPILE=/opt/riscv/bin/riscv64-unknown-elf-
export PLATFORM=qemu-riscv64-virt
export DEMO=linux+freertos
make PLATFORM=qemu-riscv64-virt DEMO=linux+freertos run
```

Here`s my tools and environment:
```
On Ubuntu 22.04

uname -a
Linux lqt-ubt 5.15.0-60-generic #66-Ubuntu SMP Fri Jan 20 14:29:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

./riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-elf/12.1.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/lqt/developtools/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g1ea978e3066 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/home/lqt/developtools/riscv-gnu-toolchain/gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.1.0 (g1ea978e3066)

```
[image]

[image]

qemu reminds me that `char device redirected to /dev/pts/1 (label serial3)`. But no output from it.

Could you please provide some info about what\`s happening here? Did i miss something important? 

Thanks a lot :)

ninolomata, 2023-07-20:
Hi @GooTal,
Thanks for reaching out!
I couldn't locate version 7.2.5 of QEMU (the version you are using) on the official repository. I successfully tested the demo on QEMU version 7.2.0 without any problems. Therefore, consider a downgrade from version 7.2.5 to 7.2.0, as it might resolve your issue.
Thank you :)

GooTal, 2023-07-20:
Hi, i cloned qemu repo, stable-7.2 branch. Then built it.
```
git clone git@github.com:qemu/qemu.git -b stable-7.2
mkdir build_riscv && cd build_riscv
../configure --target-list=riscv64-softmmu --disable-werror --enable-debug --enable-slirp
make -j8 && make install
```
The version i built is 7.2.4. It seems that 7.2.0 couldn\`t be built now. And i didn\`t notice other branches.

It printed some log here. But no more output.
[image]

Linux is fine from /dev/pts/5. But i can\`t send message by `echo hello > /dev/baoipc0`. It\`s stucked as show above.

Thanks.

josecm, 2023-07-20:
@GooTal I'm not completely sure what the issue might be :thinking: On my machine, running this on qemu 7.2.4 seems to work fine. Doesn't FreeRTOS keep printing "Task 1: ..\n Task 2: ..\n" continuously? Or does it stop after the first iteration?

GooTal, 2023-07-21:
It stopped after the first iteration. Just like the picture shown above. Only:
```
Task1: 1
Task2: 1
```

Oh, by the way, i modified the source code of wrkdir/srcs/bao/src/arch/riscv/sbi.c.
```
struct sbiret sbi_ipi_handler(unsigned long fid)
{
    if (fid != SBI_SEND_IPI_FID) return (struct sbiret){SBI_ERR_NOT_SUPPORTED};

    unsigned long hart_mask = vcpu_readreg(cpu()->vcpu, REG_A0);
    unsigned long hart_mask_base = vcpu_readreg(cpu()->vcpu, REG_A1);

    struct cpu_msg msg = {
        .handler = SBI_MSG_ID,
        .event = SEND_IPI,
    };
    fence_ord_write(); /////////////////////////////////Added this line.
    for (size_t i = 0; i < sizeof(hart_mask) * 8; i++) {
        if (bitmap_get((bitmap_t*)&hart_mask, i)) {
            vcpuid_t vhart_id = hart_mask_base + i;
            cpuid_t phart_id = vm_translate_to_pcpuid(cpu()->vcpu->vm, vhart_id);
            if(phart_id != INVALID_CPUID) cpu_send_msg(phart_id, &msg);
        }
    }

    return (struct sbiret){SBI_SUCCESS};
}
```
I added the code `fence_ord_write`. Without this line, my compiler would report 
[image]

It\`s quite strange and i think it might be caused by compiler optimization level. Adding a fence instruction causes no more errors.

So, I\`m not sure whether this line of code caused this bug. Thanks for your time. :)

josecm, 2023-07-21:
> It stopped after the first iteration.

That seems to indicate something is wrong with the FreeRTOS VM timer :thinking: 

> gcc version is 12.1.0 .
> 
> It`s quite strange and i think it might be caused by compiler optimization level. Adding a fence instruction causes no more errors.

This is very likely. We are still using GCC 10 for risc-v. Can you give me instructions on how to get/build the compiler version you are using so I can replicate the issue here?

> Thanks for your time. :)

Always glad to help!

GooTal, 2023-07-21:
Oh, i\`m not quite famaliar with riscv toolchain. I remember cloned the repo and built it. I notice -v could provide me some info. 
```
$ riscv64-unknown-linux-gnu-gcc -v

Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-linux-gnu/12.1.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/lqt/developtools/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-pkgversion=g1ea978e3066 --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=/home/lqt/developtools/riscv-gnu-toolchain/gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.1.0 (g1ea978e3066)
```

Can i send it to your email?

It seems older version toolchains is only available from the [old releases.](https://github.com/riscv-collab/riscv-gnu-toolchain/tags). I downloaded gcc version 10. But it reported errors about hard float and soft float linking errors. So still used the toolchain i\`ve built.

ninolomata, 2023-07-24:
Hi @GooTal,

In this print, you show the Linux toolchain, but you should use the baremetal toolchain to compile Bao. 
I tested compiling Bao with gcc 12.1.0 and realized there is an issue when compiling with optimizations level 2. I opened an issue in [#78](https://github.com/bao-project/bao-hypervisor/issues/78).
Until we resolve it, you have three options:

- compile Bao with optimizations level 0 or 1;
- suppress that error into a warning;
- use gcc version 10 -> link here [toolchain](https://static.dev.sifive.com/dev-tools/freedom-tools/v2020.08/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14.tar.gz).

Thank you.

GooTal, 2023-07-24:
OK, Thanks. I\`ll follow this problem in the new issue you opened.

ninolomata, 2023-08-22:
Hi @GooTal,

This issue was fixed in [#79](https://github.com/bao-project/bao-hypervisor/pull/79).
Thank you !

closed 2023-08-22 by ninolomata