bao-project/bao-demos #39: Cannot access Linux host through ssh FreeRTOS+Linux

e-nieto, 2023-07-25:
Good evening team!

I've been trying to access the Linux guest OS through ssh according to the Linux+FreeRTOS guide (IP 192.168.42.15).

The system is able to boot according to FreeRTOS on UART:
[image]

When trying to ping or ssh, host appears offline (100% packet drop, not even connection refused). However, nmap is able to detect the RPi4 as being online and being able to connect, but all ports are ignored:

[image]

I am on a fresh make from commit e6f1f6f

josecm, 2023-07-25:
@e-nieto Now that I've checked the readme files and the steps printed out when running make, I realize that something is not clear:  the demo requires you to set up your machine's eth interface IP statically to something in the same private network. For example, 192.168.42.2/24. 

Have you done this?

e-nieto, 2023-07-25:
No I didn't, but makes sense. I'll try tomorrow and report back :)

e-nieto, 2023-07-26:
So i tried to configure it into my home router, and the address space for it lays under a complete different subnet. Due to my router's software I couldn't assign it .42.15 as static address without messing up my complete home network.

I tried to configure a DHCP server under Debian but for whatever the reason it doesn't want to start a server for the ethernet interface. I'd call this one a "skill issue" on my side (I hate networking). Maybe a couple of pointers into this one helps with allowing an ad-hoc without the need to mess with the home router.

Another question would be: is there a way to configure this static IP through config.txt or other files in the boot directory? Or does it need to be done at compile time? If so I'd like to explore this venue to be able to integrate it on the home network as well with the appropriate subnet :)

Thank you!

josecm, 2023-10-21:
I'm guessing @yanjiew1 wrongly marked this issue on their PR since this is related to the Linux+FreeRTOS demo, not the Linux+Zephyr.

@e-nieto sorry for not taking a look at this sooner. Were you able to change the static IP? I believe this would be possible by chaing the `ip` parameter in the kernel command line defined in the device tree's `chosen/bootargs` property.

GOAguiar99, 2024-03-08:
Hello,

I've encountered an issue similar to the one previously reported by @e-nieto. Could anyone provide insights into the possible causes of this problem?

Thank you.

PR #46 (2023-10-21): Update zephyr to 3.5.0-rc1 and add support for linux+zephyr on rpi4
The pull request updates zephyr to version 3.5.0-rc1, which supports the soc, bcm2711, and rpi4 board. This is the prerequisite for supporting linux+zephyr on rpi4.

In addition to the update of zephyr, also add support for linux+zephyr on rpi4 in another commit.

This PR already contains the changes in the PR #45.

Closes #39
Closes #44