bao-project/bao-demos #42: What is the difference between 'bao' and 'jailhouse'

fangkaix, 2023-09-12:


sandro2pinto, 2023-09-12:
You may want to take a look at one of our recent academic papers.
"Shedding Light on Static Partitioning Hypervisors for Arm-based Mixed-Criticality Systems"
https://arxiv.org/abs/2303.11186

josecm, 2023-09-12:
@fangkaix Despite Bao being inspired by Jailhouse and following the static partitioning idea, they are completely separate implementations. The main difference between the two is that Jailhouse requires the Linux privileged "root cell" to boot and launch the other cells, while Bao is self-sufficient and has no privileged guests and boots directly after the boot loader. While it allows it to be more dynamic (starting, pausing, or stopping new cells while the system is running), this comes with the big downside of a larger system TCB and boot times. Also, you must always have that Linux guest while you can have a configuration in Bao without any Linux guest.

There is also a difference in the platforms and architecture supported. Both Bao and Jailhouse support Arm-A, but only Bao supports Armv8-R; both support RISC-V (still not upstream for Jailhouse I believe), and only Jailhouse supports x86/64 systems.

As mentioned in the paper Sandro references above, you will find other small differences regarding performance, latencies, and how some features are implemented (e.g. cache coloring and inter-vm communication are implemented in different ways) and provide different interfaces and capabilities).