bao-project/bao-demos #56: Can Bao support Raspberry Pi 5 8GB

AriSan-XD, 2024-06-15:


Diogo21Costa, 2024-08-22:
Hi @AriSan-XD ,

Thank you for your interest in using Bao on the Raspberry Pi 5 8GB! Currently, Bao does not officially support the Raspberry Pi 5. However, there is existing support for the Raspberry Pi 4, and given the similarities between the two platforms, porting Bao to the Raspberry Pi 5 might be relatively straightforward.

The process of porting the Bao hypervisor essentially involves two key steps:

1. Creating a Platform Description Folder: You would need to create a new directory under src/platform to hold the platform definition source and header files. For instance, to port Bao to a new platform named "foo," you would create the directory src/platform/foo.
2. Adding the Necessary Files: Within this directory, you would include the following four files:
    1. A source file, such as [rpi4_desc.c](https://github.com/bao-project/bao-hypervisor/blob/main/src/platform/rpi4/rpi4_desc.c), which contains the platform description structure (`struct platform platform` definition).
    2. A [platform.h](https://github.com/bao-project/bao-hypervisor/blob/main/src/platform/rpi4/inc/plat/platform.h) header file, which includes platform-specific definitions and headers.
    3. A [platform.mk](https://github.com/bao-project/bao-hypervisor/blob/main/src/platform/rpi4/platform.mk) makefile that defines necessary build system parameters, such as the target architecture.
    4. An [objects.mk](https://github.com/bao-project/bao-hypervisor/blob/main/src/platform/rpi4/objects.mk) file that lists the platform's target object files.

If you're interested in contributing to this effort, we would appreciate your help. Contributions are always welcome, and we're available to provide any guidance or support you might need.

Feel free to reach out if you have any questions or need assistance getting started.

knucklehead96, 2025-04-04:
Hi @Diogo21Costa, I am trying to port Bao to rpi5, 

can you help me in understating how the the address mapping is done in 

for example in rpi4

    .console = {
        .base = 0xfe215000,
    },

    .arch = {
        .gic = {
            .gicd_addr = 0xff841000,
            .gicc_addr = 0xff842000,
            .gich_addr = 0xff844000,
            .gicv_addr = 0xff846000,
            .maintenance_id = 25,
        },
    },

Diogo21Costa, 2025-05-02:
Hi @knucklehead96, sorry for the late response. To determine the `.console.base` address for UART1 on the Raspberry Pi 4B, it's essential to understand the BCM2711 SoC's address mapping.

### Address Mapping Overview

1. **Legacy Master View**:
   - In the BCM2711 Technical Reference Manual (TRM), the legacy master view maps peripherals starting at `0x7C000000`.
   - Within this view, UART1 is located at `0x7E215000`.

2. **ARM View in Low Peripheral Mode**:
   - By default, the Raspberry Pi 4B boots in *Low Peripheral Mode*, where the ARM cores access peripherals starting at `0xFC000000`.
   - In this mode, the legacy peripheral addresses (`0x7E000000` range) are translated to the `0xFE000000` range for the ARM cores.
   - Therefore, UART1's address `0x7E215000` becomes `0xFE215000` from the ARM's perspective.

This translation is detailed in the BCM2711 ARM Peripherals datasheet:

> "So a peripheral described in this document as being at legacy address 0x7Enn_nnnn is available in the 35-bit address space at 0x4_7Enn_nnnn, and visible to the ARM at 0x0_FEnn_nnnn if Low Peripheral mode is enabled."  
> — [BCM2711 ARM Peripherals Datasheet](https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf?utm_source=chatgpt.com)

When configuring the `.console.base` in Bao or similar systems, use `0xFE215000` to access UART1 from the ARM cores.

### GIC Configuration

Following Section 6.5.1 of the BCM2711 TRM, the GIC-400 interrupt controller has an ARM-only base address of `0x4C0040000`. When Low Peripheral mode is enabled, this base address becomes `0xFF840000`.

For full register details, please refer to the [ARM GIC-400 documentation](https://developer.arm.com/documentation/ihi0048/latest). In Table 3-1 you can find the GIC-400 memory map and in table 2-3 you can find the definition of the interrupt ID of the virtual maintenance interrupt (`.maintenance_id`).

---

Let me know if you need further assistance!