bao-project/bao-demos #59: Regarding the adaptation of BAO to the Rockchip platform

opendvd, 2025-01-03:
I'm very joyful that you and your team can open source a wonderful hypervisor framework which named Bao. I have two questions I'd like to consult you on. 

1. The first question is whether the current framework can be adapted to the Rockchip platform. 
2. The second question is, if it's not compatible with the ROCKCHIP platform, what additional work do we need to develop to make BAO compatible with Rockchip?

I'm delighted with your response. Thank you in advance.

sandro2pinto, 2025-01-03:
Thank you for the feedback and kind words @opendvd.

1. Yes, for sure. We have som work with the 96Boards ROCK960. What target Rockchip platform you have in mind?
2. Depending on the exact target SoC / platform, but in general not much. Please provide more details on (1) and then @josecm and @DavidMCerdeira can provide more details.

Thanks

opendvd, 2025-01-04:
> Thank you for the feedback and kind words @opendvd.
>
> 1. Yes, for sure. We have som work with the 96Boards ROCK960. What target Rockchip platform you have in mind?
> 2. Depending on the exact target [...]

Thank you very much for your kind reply.
I'm sorry for not providing relevant details. Here are some details about Rockchip platform.

1. The Rockchip platform we used is RK3588 which is very similar to rk3399 (ROCK960). It's a quad-core Cortex-A76 + quad-core Cortex-A55 processor, with up to 2.38GHz frequency.
2. The interrupt controller of RK3588 utilizes GICv3.
3. The RK3588 we currently use boots the operating system from the EMMC device which has 32GB storage.

Looking forward to your reply. Thank you in advance.

DavidMCerdeira, 2025-01-04:
Greetings,

> The Rockchip platform we used is RK3588 which is very similar to rk3399 (ROCK960). It's a quad-core Cortex-A76 + quad-core Cortex-A55 processor, with up to 2.38GHz frequency.

This is good news as we have internally experimented with an rk3399 platform (ROCK960) and ported Bao to it a few years ago.

> The interrupt controller of RK3588 utilizes GICv3.

Bao already has support for GICv3 so that will not be an issue.

> The RK3588 we currently use boots the operating system from the EMMC device which has 32GB storage.

It may be necessary to study how to fit Bao into the platforms boot flow. Typically we use U-Boot and access the sdcard. We have succeed in booting Bao in other configurations, for example by adding bao as binary to U-Boot SPL fit image.

Here's a reply that addresses what typically needs to be done to add a new platform to Bao: https://github.com/bao-project/bao-demos/issues/56#issuecomment-2304398471.

Let me know if you have any questions.

sandro2pinto, 2025-01-04:
@opendvd can you please share the exact reference of the development kit / platform (and not the Rockchip SoC) that you are targeting?

opendvd, 2025-01-11:
I'm so sorry for the long delay in replying. 
@DavidMCerdeira Thanks for your kind reply.
@sandro2pinto Our reference of the development kit / platform is: http://www.topeetboard.com/sydymfl/Product/iTOP-3588.html
More details about our kit are available on the website.

opendvd, 2025-01-18:
> This is good news as we have internally experimented with an rk3399 platform (ROCK960) and ported Bao to it a few years ago.

@DavidMCerdeira We are very happy that Bao can be deployed on the rk3399 platform. 

We are currently trying to port Bao to the RK3588 platform. As there are many similarities between RK3588 and RK3399, would it be convenient for your team to release the porting examples related to the RK3399 platform?
Thank you very much for your reply in advance.

DavidMCerdeira, 2025-01-28:
Greetings, sorry for the delay.

Here is a (very) old branch: https://github.com/DavidMCerdeira/bao-hypervisor.git
As this is very old, you should just take a look at the UART driver (not sure if current Bao version has this driver already actually) and platform description and try to incorporate it into the current version.

Let me know if there are any issues.

opendvd, 2025-02-05:
@DavidMCerdeira Sorry for the delay in replying to your message. 
Thank you very much for your respond. We'll try it based on old branch.

DavidMCerdeira, 2025-02-05:
I realize I didn't point to a branch, but the branch in question is `rock960`: https://github.com/DavidMCerdeira/bao-hypervisor/tree/rock960