bao-project/bao-demos #61: Dose BAO linux+freertos supported GDB  on arm64 virtual platform?

JosonFD, 2025-01-11:
I want to used GDB to debug VM linux,how to configure?

josecm, 2025-01-11:
Hey Joson, I assume that by "virtual platforms" you are referring to the FVP Models. In those platforms I have always used the arm debugger that comes with Arm development studio. I think the question ir more regarding the models than Bao itself. So I don't think I can help you setup GDB with the models.

JosonFD, 2025-01-12:
Hello, Josecm, thank you very much for your reply!
Currently, I have two questions:
My question is about how to use GDB to debug BAO HYPERVISOR + LINUX on the QEMU emulation platform. How should I configure it? Could you provide a corresponding tutorial repository?
I am adapting our BAO to run on the Phytium D2000(armv8 a73*8,gicv3,) hardware platform and running virtual machines with Linux + FreeRTOS. I encountered an issue where there is no output after jumping to Linux. I suspect the problem happens in the assembly head.S stage. I am currently using UART for debugging, but when entering Linux, after __enable_mmu function, the serial port stops working. When mapping the UART device address in the virtual machine, how should I configure it in stage1 page manage, especially in head.S?
Looking forward to your reply, and thank you very much!

JosonFD, 2025-01-12:
> Hey Joson, I assume that by "virtual platforms" you are referring to the FVP Models. In those platforms I have always used the arm debugger that comes with Arm development studio. I think the [...]

Hello, Josecm, thank you very much for your reply!
Currently, I have two questions:
My question is about how to use GDB to debug BAO HYPERVISOR + LINUX on the QEMU emulation platform. How should I configure it? Could you provide a corresponding tutorial repository?
I am adapting our BAO to run on the Phytium D2000(armv8 a73*8,gicv3,) hardware platform and running virtual machines with Linux + FreeRTOS. I encountered an issue where there is no output after jumping to Linux. I suspect the problem happens in the assembly head.S stage. I am currently using UART for debugging, but when entering Linux, after __enable_mmu function, the serial port stops working. When mapping the UART device address in the virtual machine, how should I configure it in stage1 page manage, especially in head.S?
Looking forward to your reply, and thank you very much!

josecm, 2025-01-14:
> Could you provide a corresponding tutorial repository?

We do not have any guides on debugging bao. But in our experience simply adding the symbol file (vmlinux, but you must compile linux with the debug options enabled).

I suspect your issue has something to do with the configuration you are giving Bao and maybe a mismatch with the device tree you are giving Linux. Could you provide these files so maybe we could help you out.

JosonFD, 2025-01-22:
> We do not have any guides on debugging bao. But in our experience simply adding the symbol file (vmlinux, but you must compile linux with the debug options enabled).
>
> I suspect your issue has [...]

Device Tree Configuration for Serial Output (PL011 UART, Base Address 0x28001000)
You want to configure your device tree to use the PL011 UART at the base address 0x28001000 for serial debugging output. Below is the device tree configuration for this purpose:
`/dts-v1/;

/ {   
	#size-cells = <0x2>;
	#address-cells = <0x2>;

	cpus {
		#size-cells = <0x0>;
		#address-cells = <0x1>;

		cpu@0 {
			compatible = "phytium,ftc663\0arm,armv8";
			device_type = "cpu";
			enable-method = "psci";
			reg = <0x0>;
		};
	};

	psci {
		compatible = "arm,psci-1.0";
		method = "smc";
	};

	memory@90000000 {
		reg = <0x0 0x90000000 0x0 0x40000000>;
		device_type = "memory";
	};

	gic: intc@29000000 {
		interrupts = <0x01 0x09 0x04>;
		reg = <0x00 0x29000000 0x00 0x60000 0x00 0x29100000 0x00 0x2000>;
		#redistributor-regions = <0x01>;
		compatible = "phytium,d2000-ixic";
		interrupt-controller;
		#interrupt-cells = <0x03>;
        interrupt-parent = <&gic>;
	};

   	timer {
		compatible = "arm,armv8-timer";
		interrupt-parent = <&gic>;
		interrupts = <0x01 0x0d 0x08 0x01 0x0e 0x08 0x01 0x0b 0x08 0x01 0x0a 0x08>;
	};

	uartclk: uartclk {
		compatible = "fixed-clock";
		#clock-cells = <0x0>;
		clock-frequency = <0x2dc6c00>;
		phandle = <0x1>;
	};

    	bao-ipc@f0000000 {
        	compatible = "bao,ipcshmem";
        	reg = <0x0 0xf0000000 0x0 0x00010000>;
		read-channel = <0x0 0x2000>;
		write-channel = <0x2000 0x2000>;
        	interrupts = <0 52 1>;
		interrupt-parent = <&gic>;
		id = <0>;
    	};

	uart@28001000 {
			compatible = "arm,pl011\0arm,primecell";
			status = "ok";
			interrupts = <0x00 0x07 0x04>;
			reg = <0x00 0x28001000 0x00 0x1000>;
			clock-names = "uartclk\0apb_pclk";
			clocks = <0x03 0x03>;
	};

	aliases {
		serial1 = "uart@28001000";
	};

	chosen {
		bootargs = "earlycon console=ttyAMA1,115200n8 clk_ignore_unused ip=192.168.42.15 carrier_timeout=0";
		stdout-path = "serial1:115200n8";
	};
};`

JosonFD, 2025-01-22:
Device Tree Configuration for Serial Output (PL011 UART, Base Address 0x28001000)
I want to configure device tree to use the PL011 UART at the base address 0x28001000 for serial debugging output. Below is the device tree configuration for this purpose:
`/dts-v1/;

/ {   
	#size-cells = <0x2>;
	#address-cells = <0x2>;

	cpus {
		#size-cells = <0x0>;
		#address-cells = <0x1>;

		cpu@0 {
			compatible = "phytium,ftc663\0arm,armv8";
			device_type = "cpu";
			enable-method = "psci";
			reg = <0x0>;
		};
	};

	psci {
		compatible = "arm,psci-1.0";
		method = "smc";
	};

	memory@90000000 {
		reg = <0x0 0x90000000 0x0 0x40000000>;
		device_type = "memory";
	};

	gic: intc@29000000 {
		interrupts = <0x01 0x09 0x04>;
		reg = <0x00 0x29000000 0x00 0x60000 0x00 0x29100000 0x00 0x2000>;
		#redistributor-regions = <0x01>;
		compatible = "phytium,d2000-ixic";
		interrupt-controller;
		#interrupt-cells = <0x03>;
        interrupt-parent = <&gic>;
	};

   	timer {
		compatible = "arm,armv8-timer";
		interrupt-parent = <&gic>;
		interrupts = <0x01 0x0d 0x08 0x01 0x0e 0x08 0x01 0x0b 0x08 0x01 0x0a 0x08>;
	};

	uartclk: uartclk {
		compatible = "fixed-clock";
		#clock-cells = <0x0>;
		clock-frequency = <0x2dc6c00>;
		phandle = <0x1>;
	};

    	bao-ipc@f0000000 {
        	compatible = "bao,ipcshmem";
        	reg = <0x0 0xf0000000 0x0 0x00010000>;
		read-channel = <0x0 0x2000>;
		write-channel = <0x2000 0x2000>;
        	interrupts = <0 52 1>;
		interrupt-parent = <&gic>;
		id = <0>;
    	};

	uart@28001000 {
			compatible = "arm,pl011\0arm,primecell";
			status = "ok";
			interrupts = <0x00 0x07 0x04>;
			reg = <0x00 0x28001000 0x00 0x1000>;
			clock-names = "uartclk\0apb_pclk";
			clocks = <0x03 0x03>;
	};

	aliases {
		serial1 = "uart@28001000";
	};

	chosen {
		bootargs = "earlycon console=ttyAMA1,115200n8 clk_ignore_unused ip=192.168.42.15 carrier_timeout=0";
		stdout-path = "serial1:115200n8";
	};
};`

JosonFD, 2025-01-22:
> We do not have any guides on debugging bao. But in our experience simply adding the symbol file (vmlinux, but you must compile linux with the debug options enabled).
>
> I suspect your issue has [...]

The oringnal device tree :
`/ {
        compatible = "phytium,d2000";
        interrupt-parent = <0x00000001>;
        #address-cells = <0x00000002>;
        #size-cells = <0x00000002>;
        model = "D2000 Development Board";
        reserved-memory {
                #address-cells = <0x00000002>;
                #size-cells = <0x00000002>;
                ranges;
                reserved@b0000000 {
                        reg = <0x00000000 0xb0000000 0x00000000 0x10000000>;
                        no-map;
                };
        };
        aliases {
                ethernet0 = "/soc/eth@2820c000";
                ethernet1 = "/soc/eth@28210000";
        };
        psci {
                compatible = "arm,psci-1.0";
                method = "smc";
                cpu_suspend = <0xc4000001>;
                cpu_off = <0x84000002>;
                cpu_on = <0xc4000003>;
                sys_poweroff = <0x84000008>;
                sys_reset = <0x84000009>;
        };
        cpus {
                #address-cells = <0x00000002>;
                #size-cells = <0x00000000>;
                cpu@0 {
                        device_type = "cpu";
                        compatible = "arm,armv8";
                        reg = <0x00000000 0x00000000>;
                        enable-method = "psci";
                        numa-node-id = <0x00000000>;
                        clocks = <0x00000002 0x00000000>;
                };
                cpu@1 {
                        device_type = "cpu";
                        compatible = "arm,armv8";
                        reg = <0x00000000 0x00000001>;
                        enable-method = "psci";
                        numa-node-id = <0x00000000>;
                        clocks = <0x00000002 0x00000000>;
                };
                cpu@100 {
                        device_type = "cpu";
                        compatible = "arm,armv8";
                        reg = <0x00000000 0x00000100>;
                        enable-method = "psci";
                        numa-node-id = <0x00000000>;
                        clocks = <0x00000002 0x00000001>;
                };
                cpu@101 {
                        device_type = "cpu";
                        compatible = "arm,armv8";
                        reg = <0x00000000 0x00000101>;
                        enable-method = "psci";
                        numa-node-id = <0x00000000>;
                        clocks = <0x00000002 0x00000001>;
                };
                cpu@200 {
                        device_type = "cpu";
[... 396 lines cut ...]
                        bus-range = <0x00000000 0x000000ff>;
                        interrupt-map-mask = <0x00000000 0x00000000 0x00000000 0x00000007>;
                        interrupt-map = * 0x00000000fae431d8 [0x00000080];
                        ranges = * 0x00000000fae43264 [0x00000054];
                };
        };
        chosen {
                bootargs = "console=ttyAMA1,115200 earlycon=pl011,0x28001000 root=/dev/sda4 rootdelay=5 rw";
                stdout-path = "uart1:115200n8";
        };
        memory@00 {
                device_type = "memory";
                reg = <0x00000000 0x80000000 0x00000000 0x7b000000>;
        };
        memory@01 {
                device_type = "memory";
                reg = <0x00000020 0x00000000 0x00000003 0x80000000>;
        };
        firmware {
                optee {
                        compatible = "linaro,optee-tz";
                        method = "smc";
                };
        };
};`

JosonFD, 2025-01-22:
> We do not have any guides on debugging bao. But in our experience simply adding the symbol file (vmlinux, but you must compile linux with the debug options enabled).
>
> I suspect your issue has [...]

The original runnable device tree is as follows.

`/ {
        compatible = "phytium,d2000";
        interrupt-parent = <0x00000001>;
        #address-cells = <0x00000002>;
        #size-cells = <0x00000002>;
        model = "D2000 Development Board";
        reserved-memory {
                #address-cells = <0x00000002>;
                #size-cells = <0x00000002>;
                ranges;
                reserved@b0000000 {
                        reg = <0x00000000 0xb0000000 0x00000000 0x10000000>;
                        no-map;
                };
        };
        aliases {
                ethernet0 = "/soc/eth@2820c000";
                ethernet1 = "/soc/eth@28210000";
        };
        psci {
                compatible = "arm,psci-1.0";
                method = "smc";
                cpu_suspend = <0xc4000001>;
                cpu_off = <0x84000002>;
                cpu_on = <0xc4000003>;
                sys_poweroff = <0x84000008>;
                sys_reset = <0x84000009>;
        };
        cpus {
                #address-cells = <0x00000002>;
                #size-cells = <0x00000000>;
                cpu@0 {
                        device_type = "cpu";
                        compatible = "arm,armv8";
                        reg = <0x00000000 0x00000000>;
                        enable-method = "psci";
                        numa-node-id = <0x00000000>;
                        clocks = <0x00000002 0x00000000>;
                };
                cpu@1 {
                        device_type = "cpu";
                        compatible = "arm,armv8";
                        reg = <0x00000000 0x00000001>;
                        enable-method = "psci";
                        numa-node-id = <0x00000000>;
                        clocks = <0x00000002 0x00000000>;
                };
                cpu@100 {
                        device_type = "cpu";
                        compatible = "arm,armv8";
                        reg = <0x00000000 0x00000100>;
                        enable-method = "psci";
                        numa-node-id = <0x00000000>;
                        clocks = <0x00000002 0x00000001>;
                };
                cpu@101 {
                        device_type = "cpu";
                        compatible = "arm,armv8";
                        reg = <0x00000000 0x00000101>;
                        enable-method = "psci";
                        numa-node-id = <0x00000000>;
                        clocks = <0x00000002 0x00000001>;
                };
                cpu@200 {
[... 398 lines cut ...]
                        interrupt-map-mask = <0x00000000 0x00000000 0x00000000 0x00000007>;
                        interrupt-map = * 0x00000000fae431d8 [0x00000080];
                        ranges = * 0x00000000fae43264 [0x00000054];
                };
        };
        chosen {
                bootargs = "console=ttyAMA1,115200 earlycon=pl011,0x28001000 root=/dev/sda4 rootdelay=5 rw";
                stdout-path = "uart1:115200n8";
        };
        memory@00 {
                device_type = "memory";
                reg = <0x00000000 0x80000000 0x00000000 0x7b000000>;
        };
        memory@01 {
                device_type = "memory";
                reg = <0x00000020 0x00000000 0x00000003 0x80000000>;
        };
        firmware {
                optee {
                        compatible = "linaro,optee-tz";
                        method = "smc";
                };
        };
};
`