bao-project/bao-demos #64: Linux+freertos configuration question on rpi4

lcothales, 2025-03-21:
Hi,

I am trying to build a bao demo on rpi4 with two linux. To begin my journey into the bao world, I ran the linux+freertos demo (by the way, thanks for these demos). It works, but I have a few questions about configuring bao for hardware devices.

1. In the bao configuration structure (rpi4.c), the network device for the guest Linux (genet) is configured with a physical address at 0xfd580000 but in the device tree (linux.dts), this device is located at 0x7d580000. How can the network work in the guest with this configuration ?

2. Concerning freertos, the gic address is set to 0xF9010000 in the bao configuration. Where does this address come from ?

Thanks in advance,

josecm, 2025-03-22:
Hello @lcothales. Thanks for experimenting with Bao for your purposes! Regarding your questiosn, please see inline:

> 1. In the bao configuration structure (rpi4.c), the network device for the guest Linux (genet) is configured with a physical address at 0xfd580000 but in the device tree (linux.dts), this device is located at 0x7d580000. How can the network work in the guest with this configuration ?

The rpi4 has a very suis generis address space. There is not a global physical address space and different masters may observe a peripheral in different addresses. This happens for the CPUs. Please take a look at section "1.2. Address map" of the [BCM2711 Arm Peripherals document](https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf). In the device tree this "translation" if indicated by the *ranges*  field in the scb bus node. Please take a look at official device tree specification for a better understanding of how this works.

> 2. Concerning freertos, the gic address is set to 0xF9010000 in the bao configuration. Where does this address come from ?

Our freertos port sits upon the baremetal guest enviroment we have developed. Although if you build this baremetal enviroment for a given platform it will use real addresses for that platform (for memory and peripherals such as the GIC), it also supports the idea of a "standard address space" so we can more easily share similar configurations for different pltaforms. Building it with `STD_ADDR_SPACE=y' will compile the baremetal enviroment for fixed addresses which are based on the zcu104 (the first platform bao was developed for). In the demos we are doing exactly this. Thus that address is originally the address of the GIC in the ZCU104.

Hope this answers your questions

lcothales, 2025-03-26:
Thanks a lot, very clear answer.

closed 2025-03-26 by lcothales